SwerveRobotics/ftc_app #26: Small Documentation Error

nitrocalcite, 2015-09-27:
Hello!  First I wanted to say how awesome the features are in here!  Being a rookie team with no experience, we had some issues using I2C sensors and this fixed them in addition to providing a lot of nice features.  There's just a small problem in the documentation - under IBNO055IMU.Position, the x, y, and z vars are described as velocity, where I think they should be position.  It may also be helpful to clarify what sort of position - it's relative to the location where the position routine is started up, right?  So (0,0,0) is the robot start location?

rgatkinson, 2015-09-27:
Thank you both for your note and for taking the time to explore our library. We've fixed this; you'll see it the next time we push to the #master branch. You are correct about position (and velocity). You specify initial values when you start the integrator, and thereafter we just report what the integrator has computed given acceleration stimuli. Given that the accelerations in question are the accelerations *of the sensor*, it makes sense that the initial velocity at least be that of the sensor (usually zero). Depending on the geometry of the robot and the needs of your code, it might be convenient to include in the initial value a fixed position offset from the sensor to some more noteworthy location on the robot, perhaps its center of rotation.

rgatkinson, 2015-09-27:
Also: in the next push you'll also see we've made the integration algorithm pluggable. What's there by default does the basic math, but isn't very good. To use on a bot, a more sophisticated algorithm will reasonably be needed. Filtering out noise seems important, for example; there's lots out there on the net about this.

rgatkinson, 2015-09-27:
Fixed.

closed 2015-09-27 by rgatkinson

nitrocalcite, 2015-09-27:
Thanks so much!