SwerveRobotics/ftc_app #46: Can't switch to configuration or logging activity

IronReignRobotics, 2015-11-02:
We updated to your latest and can no longer configure the robot through settings.  We also can't view logs or view the about page.  It visually looks like the new activity is killing itself resulting in a reload of the underlying ftcrobotcontroller activity which tries to restart the robot.

Could this have something to do with commit 391a1d4  I agree with the reasoning for the auto-kill feature - it's driving us crazy every time controllers go unstable - it's best to force the app to close.

rgatkinson, 2015-11-02:
While that’s certainly sounds eminently plausible, given your description, we’re unable to reproduce this on our side.

A couple of questions. Do you get a behavioral change if you comment out the android.os.Process.killProcess line in FtcRobotControllerActivity.java? Does it make a difference if you entirely uninstall the FTC Robot Controller app from the RC phone, and then retry?

pondersome, 2015-11-02:
I did try commenting out the whole onDestroy() override and that stopped the problem from happening.  I'll try just commenting out the killProcess and let you know.  Right now it seems like onDestroy is firing whenever the menu switches to a different activity.

pondersome, 2015-11-02:
OK, I retract the issue.  If I comment out the kill line, I do get the logcat message indicating process suicide, so the onDestroy was getting called within a second or two on any switch to another activity.  The reason, though, is that the "Don't keep activities"  Developer option was checked on our phone. That's what was causing the destroy event to fire aggressively. I have no idea why that option was checked - it's not checked on any of our other phones. When unchecked the problem goes away.  Thanks again for sharing your extensions to ftc_app - there's so  much great work in there.

pondersome, 2015-11-02:
Sorry, I realized my github login changed while adding these latest responses.  I'm a coach for Iron Reign and originally reported the issue under the IronReign admin account.

pondersome, 2015-11-02:
One last comment - with that debug option unchecked and the kill in place, it doesn't seem like the app fully closes any earlier.  It still requires an explicit swipe from the list of background activities if we want a quick cycle time.  At least on our phone. Maybe onStop would get triggered earlier?

closed 2015-11-02 by IronReignRobotics