SwerveRobotics/ftc_app #56: Word Read/Write in I2cDeviceClient?

luciengaitskell, 2015-11-14:
Is there any capability for word read and writes in the I2cDeviceClient class. If not, is there any planned?

rgatkinson, 2015-11-15:
We’ve thought about this, off and on, but have yet to be compelled ☺. Do you happen to have a motivating example?

luciengaitskell, 2016-01-09:
I have a MaxSonar Ultrasound sensor with requires a word read. Also, I have a led which changes color using three bytes. Is there any way to work with these with the current library?

rgatkinson, 2016-01-09:
Sure. The easiest way is to read the bytes as a byte array, then use a ByteBuffer (I think it’s called; I’m away from my main machine at the moment) to convert that to an integer, either little endian, or big endian, as appropriate.

luciengaitskell, 2016-01-09:
Another question, what do I put in for the "OpMode context" in the "ClassFactory.createI2cDeviceClient()"?

rgatkinson, 2016-01-09:
Just pass the instance of the (subclass of) OpMode that you are running.

luciengaitskell, 2016-01-09:
Ok. Why is it needed?

luciengaitskell, 2016-01-09:
I'm trying to use multiple sensors on the same port using your I2cDeviceClient library, yet I seems to hang the OpMode when I try to use it

rgatkinson, 2016-02-02:
Using multiple sensors on one port is not something that is supported. Is that important to you? Can you help us understand by illustrating with an example?

closed 2016-03-05 by rgatkinson