SwerveRobotics/ftc_app #58: SyncServo broken

mig7999, 2015-11-29:
Servo usage using the sync version is broken. it only takes parameters .99 and 1.0 and goes from one side to the other without stopping in the middle at all. No other paramters values are allowed. Went back to an old version of this package and worked perfectly. The standard  RobotFTCController version works fine with servos.

rgatkinson, 2015-11-29:
Thanks for the bug report. Is your code somewhere on the net where we could try it out? Thanks…

mig7999, 2015-11-29:
Hi there. I just used the demo file "SynchServoDemo.java" and it does not
work. When i tried K9TankDrive, the servos work again. I had good luck
going back to an old version, about 6 weeks old. I am trying 1.0 right now
to see where the bug started.
Thank you for such amazing software.!
Thanks for the bug report. Is your code somewhere on the net where we could
try it out? Thanks…

https://github.com/SwerveRobotics/ftc_app/issues/58>.

rgatkinson, 2015-11-29:
Thanks. We’ll have a look.

https://github.com/SwerveRobotics/ftc_app/issues/58>.

mig7999, 2015-11-29:
Indeed, the problem goes away when i revert to version 1.0.

rgatkinson, 2015-11-29:
I believe I've now found the problem and pushed a fix to the #master branch.

closed 2015-11-29 by rgatkinson