SwerveRobotics/ftc_app #62: Latest update says Incompatible apps

crowcasso, 2016-01-20:
After updating to the latest version of Swerve, our robot controller now reports "Incompatible apps: v1 and v2". In Settings --> Library Version it now reports "developer_build" instead of 11.15.04. Are these related?

cmacfarl, 2016-01-20:
You need to side load the DS.  It's in docs/app/....

Craig

closed 2016-01-20 by crowcasso

higgs241, 2016-02-05:
Do we have to worry about event people side loading fresh versions from FIRST that are incompatible with the swerve app at events?

crowcasso, 2016-02-08:
This should not happen.  The software inspectors should be okay with swerve software as long as your setup works properly. The software inspectors look for Android version, name of phones, if there is a direct wifi connection, airport more, etc. They do not look directly at the software. 

-joel