SwerveRobotics/ftc_app #63: Incompatible apps: v2 vs v1

Ace314159, 2016-01-21:
I updated my fork of the code with the new code in master, but when I try to start the robot, I get the error: Incompatible apps: v2 vs v1. I even tried installing a new and unchanged copy of the app by downloading the app at the current moment, but that still gives the same error. Is this error supposed to happen? How do I fix it? Thank you!

Ace314159, 2016-01-21:
I didn't realize the Driver Station APK inside the doc\apk\ folder. I installed the Driver Station app on the Driver Station phone and then it started to work.

closed 2016-01-21 by Ace314159

higgs241, 2016-02-05:
Sorry but that isn't a valid solution because events are liable to hotflash new code onto your phones

Ace314159, 2016-02-05:
What........I installed the apk from doc/apk into the Driver Station onto the Driver Station phone instead of the one in the Google Play Store and it worked......Why isn't that valid? Could you please explain in simpler terms? Sorry for any inconvenience I may cause.

eukota, 2016-02-05:
We have three teams that utilize the side-load from doc/apk using adb. If you download new code from the same repository, it will still work. I think you are good to go.

higgs241, 2016-02-05:
Not sure but there have been very aggressive robot/field inspectors who have just grabbed my phone and side loaded the drive station app.

Ace314159, 2016-02-05:
You could just reinstall it on your phone after inspection.

rgatkinson, 2016-02-06:
For the moment, sideloading is the only viable solution; apologies.

FTAs should not be forcibly changing driver station apps: at any given moment, the FTC HQ will have (of course) an official current release, but possibly others in beta (this will be more common going forward), any of which are reasonable for teams to base their code on. Generally speaking, the driver stations used in such beta releases may be incompatible with other releases, and visa-versa. Whatever code a team is using, they MUST use a matching driver station app (going forward, this 'matching' version number is the Robot Wifi Protocol Version visible on the about screen starting in the next update).

I will have a word with 'folk at FIRST, to try to get this message out to the FTA community.

higgs241, 2016-02-06:
Awesome sauce. No like I really think this should be fine but I've the unpleasant experience of an FTA just reinstalling the software on our phones. We could totally just re-sideload the driver station app but that seems a bit cloak-and-dagger and I'd rather just know and do it the legal way

rgatkinson, 2016-02-06:
'Hear you. I'm working on getting this straightened out through official channels; I'm optimistic of that being clarified relatively soon.