SwerveRobotics/ftc_app #65: NullPointerException using ClassFactory.createEasyMotorController()

Ace314159, 2016-01-30:
I called `ClassFactory.createEasyMotorController(this, rightWheel, leftWheel);`, but I get a `NullPointerException`:

`01-29 19:49:10.152 3504-4489/com.qualcomm.ftcrobotcontroller E/RobotCore: org.swerverobotics.library.internal.EasyModernMotorController.arm(EasyModernMotorController.java:128)
01-29 19:49:10.152 3504-4489/com.qualcomm.ftcrobotcontroller E/RobotCore: org.swerverobotics.library.internal.EasyModernMotorController.create(EasyModernMotorController.java:57)
01-29 19:49:10.152 3504-4489/com.qualcomm.ftcrobotcontroller E/RobotCore: org.swerverobotics.library.ClassFactory.createEasyMotorController(ClassFactory.java:73)`

This does not work with any motors that are not null.

rgatkinson, 2016-02-01:
Could you perhaps share a copy of your FtcRobotControllerActivity.java file so we could have a look? It should be the same as the one in our 'master' branch; that's different from the one in the raw SDK.

Ace314159, 2016-02-02:
Thank you! My FtcRobotControllerActivity.Java is not updated. I used the file in the 'master' branch. I will test it in the next few days. My team has switched from the `LinearOpMode` to the `SynchronousOpMode` so this issue will not be affecting us (I will still test to see if it is fixed). However, we cannot get `GamePads` and `TelemetryDashboardAndLog` to work before the `waitforStart()` in the `main()` function.

rgatkinson, 2016-02-02:
Generally, it's a lot more reliable to use git to handle the various library updates and changes rather than copying things by hand. As you've noticed, it's easy to miss things when doing things that way. If you need help with bending git to your will in this way, we can help you with that.

Ace314159, 2016-02-02:
Actually, I did use Git to update but it seems that I must not have updated properly..... Previously, I ran a `git pull`. What command should I use to accurately update?

Ace314159, 2016-02-02:
`TelemetryDashboardAndLog` works in the Synchronous OpMode during the init if I use `updateNow()`! Gamepads also work.

closed 2016-02-02 by Ace314159