SwerveRobotics/ftc_app #67: ColorSensor issues with 1.5.2

crowcasso, 2016-02-03:
Using the modern robotics color sensor causes the following error now:

ClassCastException - com.qualcomm.hardware.modernrobotics.ModernRoboticsI2cColorSensor cannot be cast to com.qualcomm.robotcore.hardware.Engagable.

The code was good before updating today. 

Any ideas?  Thanks!

Ace314159, 2016-02-04:
I also have gotten this error.

crowcasso, 2016-02-04:
Looks like we might be missing:  com.qualcomm.robotcore.hardware.usb.RobotUsbModule

This gives me an error in II2cDeviceClient.java  (which also extends Engagable).

rgatkinson, 2016-02-04:
I'll look into this; 'pologies for the inconvenience.

Ace314159, 2016-02-04:
Does the commit 2b8e462d5d195f5b286233377457bc42f3cd1ecd fix this bug?

rgatkinson, 2016-02-04:
It should, yes.

Does the commit 2b8e462<https://github.com/SwerveRobotics/ftc_app/commit/2b8e462d5d195f5b286233377457bc42f3cd1ecd> fix this bug?

crowcasso, 2016-02-04:
I am no longer having the issue after the fix mentioned above. 

Thanks for the response and make this code available to us.

closed 2016-02-04 by crowcasso