SwerveRobotics/ftc_app #68: BNO055: failed to retrieve chip id

crowcasso, 2016-02-04:
With the latest updates, our BNOO55 IMU no longer will initialize properly. Maybe we are doing something wrong.  The error occurs when we press START after pressing INIT. We have displayed in the past a message that told us the IMU was ready to go (this was to mitigate the return 0 issue mentioned in another issue). Now it returns nothing and when we try to start, after waiting a long time, we get this message:

Error: User code threw an uncaught exception: BNO055InitializationException - BNO055: failed to retrieve chip id

Code:

```
    private IBNO055IMU imu;
    private IBNO055IMU.Parameters parameters = new IBNO055IMU.Parameters();

    protected void main() throws InterruptedException {

        // setup other things

        parameters.angleUnit = IBNO055IMU.ANGLEUNIT.DEGREES;
        parameters.accelUnit = IBNO055IMU.ACCELUNIT.METERS_PERSEC_PERSEC;
        parameters.loggingEnabled = false;
        parameters.loggingTag = "BNO055";
        imu = ClassFactory.createAdaFruitBNO055IMU(hardwareMap.i2cDevice.get("gyro"), parameters);

        // enable reporting of position using the naive integrator
        imu.startAccelerationIntegration(new Position(), new Velocity());

        // make sure the IMU is working properly
        byte value = imu.getSystemStatus();
        telemetry.addData("imu", "system status = " + value);
        if (value == 0x5) {
            telemetry.addData("imu", "system is ready");
        } else {
            telemetry.addData("imu", "RESTART NEEDED");
        }
        telemetry.update();
   }
```

We'll keep debugging, but would take any suggestions people have. 

Thanks again!

rgatkinson, 2016-02-04:
Hate it when that happens. Would you happen to have a log of a problematic run I could take a look at?

                Bob

crowcasso, 2016-02-04:
Lost the log file for when the error happened -- tried to keep app running with the error message, but it decided to restart. We can generate a log file to send to you tomorrow when we’re back at the robot. 

I tried it 4 times and it failed all 4 times. 

-joel

rgatkinson, 2016-02-04:
Damn. Just ran it four times here, and it worked four times. I hate Heisenbugs.

Some thoughts: the changes recently introduced here basically took what was the initialize() method, renamed it initializeOnce(), and wrapped it in a new initialize() method that tries initialization up to four times, each time checking that the getSystemStatus() at the end is what’s expected. If it fails after four times, it throws a (new) exception.

In addition, the delay after setting the useExternalCrystal value was upped by 40ms in response to reports that might help. FWIW, for completeness, we also removed the thread priority boost functionality.

You’re not seeing the exception. I’d be curious though, if you’re going around the loop more than once; that is, whether the failure you’re seeing is the first call to initializeOnce() or a later one. Also: a lot of the issues we’ve seen with this devices are timing related. That can sometimes be discerned by setting a breakpoint at the start of initializeOnce() and stepping through it at a moderate pace, the idea being that that allows much bigger timing windows than normal. If that works, but the full speed doesn’t it’s likely a timing issue, and then we’d just need to figure out exactly where.

‘Wish I had a more ready answer….

crowcasso, 2016-02-05:
I take it all back. I went and picked up the robot to debug and generate log files.

In the process of trying to find some version that would work I completely deleted the apps and installed older versions (which worked). Then I decided to go back to the newest version (which had the issue earlier today) ... and it worked. 

Does not make a whole lot of sense -- and is not satisfying at all, but deleting and reinstalling the apps fixed this issue. 

Again, thanks for your work and willingness to help out. 

-joel

closed 2016-02-05 by crowcasso