SwerveRobotics/ftc_app #73: FTC Driver Station stuck on "Please wait while we update the Wifi Direct settings on this device"

higgs241, 2016-02-11:
Installed the app on the robot controller phone and side loaded the driver station app, but now the driver station app is stuck on that screen.

cmacfarl, 2016-02-11:
We think this is because the DS misses a "done" event for wifi setup but otherwise the initialization is complete.  I'll wait a couple minutes and then just hit back and everything seems to be fine.

higgs241, 2016-02-11:
solution is to reset the wifi direct settings in the robot controller app

higgs241, 2016-02-22:
Never mind I'm still having this issue.

lukethecoder64, 2016-02-22:
Can confirm. Usually happens to us when changing to a wifi direct channel
that has other teams on it.

nitrocalcite, 2016-03-03:
We have this issue sometimes.  It's solved by repeatedly pressing the back button until to get to the app.  No clue why, it's really odd.

cmacfarl, 2016-03-03:
Should be fixed in the next release of the SDK.

Fromen8923, 2016-03-03:
I found a cause for this, and a workaround. If you have both the RC and DS on one phone is when I've seen it happen: when the DS is started for the first time, it configures the WiFi Direct settings, and finishes properly. Then opening the RC does the same thing: it configures properly. Opening the DS after the RC is when it gets stuck on the configuration screen. So it seems to me that when the RC does its configuration, it causes some issue that doesn't allow the DS to configure properly.

The workaround I found is to simply restart the phone. My guess is that resets all of the WiFi Direct settings, meaning the DS can configure it properly.

nitrocalcite, 2016-03-03:
That solved my problem consistently, I don't have to back button anymore.  Thanks!

rgatkinson, 2016-03-05:
fixed in 1.5.0

closed 2016-03-05 by rgatkinson