SwerveRobotics/ftc_app #80: Make error messages scrollable

strugee, 2016-02-24:
Whenever there's some sort of OpMode error (usually from the hardware) the app will output some information at the bottom of the main app screen.

Unfortunately, this text is cut off with the string `...` at the end. It'd be really useful if we could actually view all error messages at the same time. I tried to just make the error text be a child of a `ScrollView`, but unfortunately the code that does the actual truncation is inside `FtcCommon-release.aar`, so I wasn't able to complete the change. (Yet another reason FTC should be open source... :/ Anyway.)

If you want me to file this issue upstream at ftctechnh/ftc_app, let me know.

strugee, 2016-02-24:
Oh, and we're on SDK 1.3.0.

cmacfarl, 2016-02-24:
The log on the Robot Controller will usually contain the full text of the error.  adb wireless with the logs piped back into Android Studio' log viewer is your friend here.

strugee, 2016-02-25:
True. But I don't want to have to open that up only to have to scroll all the way through irrelevant Android logs, just so I can know which devices are failing. There's no reason the app shouldn't just tell you.