SwerveRobotics/ftc_app #82: Robot Controller App Does Not Run, Missing Library

nitrocalcite, 2016-03-03:
It appears that java.beans.PropertyDescriptor is not being found.  I have installed and updated all SDK packages in Android Studio.  The class is being looked for during the robot init period, and it crashes every time.  Here is the stack trace:

```
03-02 19:25:32.039 7674-7698/com.qualcomm.ftcrobotcontroller E/AndroidRuntime: FATAL EXCEPTION: FtcRobotControllerService$b
                                                                               Process: com.qualcomm.ftcrobotcontroller, PID: 7674
                                                                               java.lang.NoClassDefFoundError: java/beans/PropertyDescriptor
                                                                                   at java.lang.Class.getDeclaredMethods(Native Method)
                                                                                   at java.lang.Class.getDeclaredMethods(Class.java:656)
                                                                                   at org.swerverobotics.library.internal.Util.getDeclaredMethods(Util.java:186)
                                                                                   at org.swerverobotics.library.internal.Util.getDeclaredMethodsIncludingSuper(Util.java:206)
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.findAnnotatedStaticMethodsOfInterest(AnnotatedOpModeRegistrar.java:297)
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.processAnnotatedStaticMethods(AnnotatedOpModeRegistrar.java:264)
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.doRegistration(AnnotatedOpModeRegistrar.java:126)
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.register(AnnotatedOpModeRegistrar.java:68)
                                                                                   at com.qualcomm.ftcrobotcontroller.opmodes.FtcOpModeRegister.register(FtcOpModeRegister.java:67)
                                                                                   at com.qualcomm.robotcore.eventloop.opmode.OpModeManager.registerOpModes(SourceFile:103)
                                                                                   at com.qualcomm.ftccommon.FtcEventLoop.init(SourceFile:130)
                                                                                   at com.qualcomm.robotcore.eventloop.EventLoopManager.a(SourceFile:643)
                                                                                   at com.qualcomm.robotcore.eventloop.EventLoopManager.setEventLoop(SourceFile:614)
                                                                                   at com.qualcomm.robotcore.eventloop.EventLoopManager.start(SourceFile:546)
                                                                                   at com.qualcomm.robotcore.robot.Robot.start(SourceFile:58)
                                                                                   at com.qualcomm.ftccommon.FtcRobotControllerService$b$1.run(SourceFile:173)
                                                                                   at com.qualcomm.robotcore.util.Util.logThreadLifeCycle(SourceFile:108)
                                                                                   at com.qualcomm.ftccommon.FtcRobotControllerService$b.run(SourceFile:121)
                                                                                   at java.lang.Thread.run(Thread.java:841)
                                                                                Caused by: java.lang.ClassNotFoundException: Didn't find class "java.beans.PropertyDescriptor" on path: DexPathList[[zip file "/system/framework/android.test.runner.jar", zip file "/data/app/com.qualcomm.ftcrobotcontroller-2.apk"],nativeLibraryDirectories=[/data/app-lib/com.qualcomm.ftcrobotcontroller-2, /vendor/lib, /system/lib]]
                                                                                   at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:56)
                                                                                   at java.lang.ClassLoader.loadClass(ClassLoader.java:497)
                                                                                   at java.lang.ClassLoader.loadClass(ClassLoader.java:457)
                                                                                   at java.lang.Class.getDeclaredMethods(Native Method) 
                                                                                   at java.lang.Class.getDeclaredMethods(Class.java:656) 
                                                                                   at org.swerverobotics.library.internal.Util.getDeclaredMethods(Util.java:186) 
                                                                                   at org.swerverobotics.library.internal.Util.getDeclaredMethodsIncludingSuper(Util.java:206) 
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.findAnnotatedStaticMethodsOfInterest(AnnotatedOpModeRegistrar.java:297) 
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.processAnnotatedStaticMethods(AnnotatedOpModeRegistrar.java:264) 
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.doRegistration(AnnotatedOpModeRegistrar.java:126) 
                                                                                   at org.swerverobotics.library.internal.AnnotatedOpModeRegistrar.register(AnnotatedOpModeRegistrar.java:68) 
                                                                                   at com.qualcomm.ftcrobotcontroller.opmodes.FtcOpModeRegister.register(FtcOpModeRegister.java:67) 
                                                                                   at com.qualcomm.robotcore.eventloop.opmode.OpModeManager.registerOpModes(SourceFile:103) 
                                                                                   at com.qualcomm.ftccommon.FtcEventLoop.init(SourceFile:130) 
                                                                                   at com.qualcomm.robotcore.eventloop.EventLoopManager.a(SourceFile:643) 
                                                                                   at com.qualcomm.robotcore.eventloop.EventLoopManager.setEventLoop(SourceFile:614) 
                                                                                   at com.qualcomm.robotcore.eventloop.EventLoopManager.start(SourceFile:546) 
                                                                                   at com.qualcomm.robotcore.robot.Robot.start(SourceFile:58) 
                                                                                   at com.qualcomm.ftccommon.FtcRobotControllerService$b$1.run(SourceFile:173) 
                                                                                   at com.qualcomm.robotcore.util.Util.logThreadLifeCycle(SourceFile:108) 
                                                                                   at com.qualcomm.ftccommon.FtcRobotControllerService$b.run(SourceFile:121) 
                                                                                   at java.lang.Thread.run(Thread.java:841) 
```

rgatkinson, 2016-03-03:
When did this behavior start? There haven't been any recent releases to the library of late. What sort of things changed in your environment between the time things were working and when this started happening?

sgeffner22, 2016-03-03:
Are you adding any of your own classes or components that require java beans? Also, have you tried doing a clean build?

cmacfarl, 2016-03-03:
Yup, this happens when you add a package that the annotation software can't find.

See https://github.com/cmacfarl/ftc_app/commit/eb7495f23d02985b5876c7feae26c3a7f92afa98

Was necessitated by https://github.com/cmacfarl/ftc_app/commit/dbe9bbf84b47791deca875501195d2039b681865

rgatkinson, 2016-03-03:
Improved robustness fixed and checked in; thanks Craig!

There still does remain the issue as to what is trying to draw in java beans; that's not part of the library.

nitrocalcite, 2016-03-04:
We added a new OpMode that utilized the CSVWriter class, which we inserted.  I'm trying to track down what was deleted but can't find anything.  I've updated all the Android Studio packages.  

@cmacfarl - I don't entirely understand what you are saying, sorry!  Is simply adding the CSVWriter jar not enough?  Do we need to add some sort of `@` statement?

rgatkinson, 2016-03-05:
fixed

closed 2016-03-05 by rgatkinson

nrizzio, 2016-03-05:
@rgatkinson @cmacfarl The problem has disappeared after doing a clean build and pulling the updated changes from the Swerve library. Thanks for your help!