SwerveRobotics/ftc_app #84: I2c Reading Returns -1

luciengaitskell, 2016-03-05:
I have tried to read two different (_Blink M LED_ and _Lidar-Lite v2_) sensors using this basic code (I have been able to read and write from them using SMBus on a raspberry pi):
```
II2cDeviceClient sensor = ClassFactory.createI2cDeviceClient(this, ClassFactory.createI2cDevice(hardwareMap.i2cDevice.get("sensor")), i2cAddress, false);
sensor.engage();
Log.d("readValue", "Version Number: " + String.valueOf(ls.read8(readAddress)));
sensor.disengage();
sensor.close();
```

Yet in the log it prints:
> Address:-1

Writing does work. I have tried different memory addresses and nothing seems to be working.

rgatkinson, 2016-03-06:
What value(s) are you passing for the i2cAddress parameter?

rgatkinson, 2016-03-06:
In the FTC SDK you need to take the address on the datasheet / whatever you used in Arduino or Rasberry Pi and multiply it by two.

rgatkinson, 2016-03-06:
Example: the BlinkM data sheet lists its address as 0x09. So you should pass 0x12.

luciengaitskell, 2016-03-06:
I was passing `0x00` (general address) to `i2cAddress` as `0x09` wasn't working. Thank you for telling me that. It was successfully writing but not reading.

rgatkinson, 2016-03-06:
You're welcome. I agree that it is confusing.

luciengaitskell, 2016-03-06:
Do you know how to fix the reading error though?

closed 2016-03-06 by luciengaitskell

cmacfarl, 2016-03-07:
I happen to know the guy who developed the BlinkM and passed this on to him as a fun, "Hey look your device is being used in FTC" sort of thing and he responded that, yes, indeed reads don't work through the broadcast address, which when you think about it makes a certain amount of sense.  If you have N devices on the bus which one is going to respond with your data?