SwerveRobotics/ftc_app #85: Cannot Resolve Symbol 'I2cDeviceSynch'

luciengaitskell, 2016-03-06:
I just pulled from this repository to my fork of ftctechnh/ftc_app. The `II2cDeviceClient` class is complaining that it: `Cannot Resolve Symbol 'I2cDeviceSynch'`. How would I fix this?

rgatkinson, 2016-03-06:
You need to 'import com.qualcomm.robotcore.hardware.I2cDeviceSynch;'

Odd that this isn't happening automatically, though. Have you rebuilt since you pulled?

luciengaitskell, 2016-03-06:
Didn't rebuild my project... Whoops, thanks again!

sgeffner22, 2016-03-06:
I'm experiencing a similar problem in Darrell's swerveteams repo. Android Studio isn't finding robotcore in 'import com.qualcomm.robotcore.hardware.I2cDeviceSynch'. Did a clean build on each project but still isn't resolved. Could our build.gradle be missing something? It has these dependencies:

dependencies {
    compile files('libs/android-support-v4.jar')
    compile (name:'RobotCore-release', ext:'aar')
    compile (name:'Hardware-release', ext:'aar')
    compile (name:'FtcCommon-release', ext:'aar')
    compile (name:'ModernRobotics-release', ext:'aar')
    compile (name:'Analytics-release', ext:'aar')
    compile (name:'WirelessP2p-release', ext:'aar')
    compile project(':SwerveRoboticsLibrary')
}

rgatkinson, 2016-03-06:
Is there a repo/branch i could sync to to investigate?

sgeffner22, 2016-03-06:
https://github.com/eukota/ftc_app     the SwerveTeams branch

sgeffner22, 2016-03-06:
whoops, let me commit my changes ... which will break everyone for a little while...

sgeffner22, 2016-03-06:
ok, it's ready

rgatkinson, 2016-03-06:
I cloned that, imported into Android Studio, and it built cleanly. Ugh.

rgatkinson, 2016-03-06:
Oh! Wrong branch...wait a sec...

rgatkinson, 2016-03-06:
This is a stinker!

rgatkinson, 2016-03-06:
I'm beginning to suspect it's an issue with the .iml files being in the SCC tree.

rgatkinson, 2016-03-06:
Nope. Not that: they're not checked into the tree (which is correct). It's not parsing the imports correctly for some dang reason. I once simplified things enough to get that working, but now need to make that reproducible.

rgatkinson, 2016-03-06:
Ok. Fixed and pushed. It was an issue in AndroidStudio / gradle: we tickled a bug of theirs. Updating to track the breaking changes in the release seems to have fixed things.

cmacfarl, 2016-03-06:
Where'd you push this to?  I can't find any commit(s) from today in any of your branches.

rgatkinson, 2016-03-06:
It was our internal team's code; there were no changes in the Swerve Library itself. Rather, I updated our internal code to track 1.5.0, and the dependency analysis bugs we apparently were tickling in Android Studio went away.

sgeffner22, 2016-03-06:
Awesome, much thanks Bob! Works great now. 
Also, thanks for migrating the current sensor code to 1.5 for me!

closed 2016-03-06 by sgeffner22