SwerveRobotics/ftc_app #92: February SDK update not in this repository

nrizzio, 2016-05-15:
83d998b9e9937e51b96acae151aa10b993ca1a83 has been on the upstream repository for almost three months now but it hasn't been merged into Swerve's master branch (based on the "1 commit behind of ftctechnh/ftc_app hint on the main page).

@rgatkinson

nrizzio, 2016-05-22:
Actually, I'm (mostly) wrong. [It uses the March beta libraries](https://github.com/SwerveRobotics/ftc_app/blob/master/README.md#how-to-use):
> It currently is synchronized to the beta release from FTC HQ that was published in March, 2016 (version 1.7).

@rgatkinson, I would prefer that the libraries here would be updated by `git merge`ing from FIRST's repository (like [this merge commit](https://github.com/SwerveRobotics/ftc_app/commit/c34f76e425962e3349bc12b1f19f7360260b2c73)), since it makes for a cleaner git history and prevents the following from happening (and only getting worse) on the main repository page:
[image]
The reason being that this can be a great visual indicator of whether new changes have yet to be introduced into this SDK.

Merge conflicts will be detected and can be resolved manually if they happen, which I think they will if you merge the latest commit from the FTC beta branch.

But that's just my opinion. If anyone else has any comments, please post!

closed 2016-05-22 by nrizzio

rgatkinson, 2016-05-23:
@nrizzio: While I understand your point, and it makes sense, it turns out to be surprisingly difficult to achieve in practice due to the the multiple versions of multiple libraries that we're maintaining. Apologies, but I don't think we can accommodate this.

One thing though: we do *not* expect that anyone will merge the FTC tree into the Swerve tree as you allude to. That's not something we support, as, generally, changes in the core SDK require concomitant changes in the Swerve tree (though that is decreasing over time). We do try hard to minimize any lag involved between the releases.