SwerveRobotics/ftc_app #93: Execution failed

TeameurekaRobotics, 2016-06-19:
Can you help explain this error? I have synced with the latest Swerve code from GitHub. 

Error:Execution failed for task ':SwerveRoboticsLibrary:compileReleaseJavaWithJavac'.
> java.io.FileNotFoundException: C:\Users\hsrobotics\Documents\GitHub\My_Swerve_ftc_app\SwerveRoboticsLibrary\libs\android-support-v4.jar (The system cannot find the path specified)

nrizzio, 2016-06-19:
Did you update the Gradle plugin to the most recent version (as Android Studio suggests you do)?

TeameurekaRobotics, 2016-06-19:
To the best of my knowledge. How would one go about checking?

TeameurekaRobotics, 2016-06-19:
I followed these instructions for Gradle, and it doesn't indicate an update.

[]([](https://www.jetbrains.com/help/idea/2016.1/installing-updating-and-uninstalling-repository-plugins.html?origin=old_help&search=gradle%20plugin%20update) )

NoahAndrews, 2016-06-20:
Make sure you have the latest version of the Android Support Repository installed in the Android SDK Manager.

TeameurekaRobotics, 2016-06-20:
Android Support SDK shows no updates. rev33

cmacfarl, 2016-06-20:
http://ftcforum.usfirst.org/showthread.php?6738-Android-Studio-2-0-and-SDK-Release-16-03-09-Signed-Gradle

TeameurekaRobotics, 2016-06-20:
OK, then. Well, I'm not sure if I want to say thanks, due to the rabbit hole this post thread took me down. However, in the end I appear to have fixed the error. I just want to add here, that this is ridiculous for the level of complexity a team with limited Android Studio abilities is needing to simply make things work.

From that post I was able to discern several things: One; the issue has something to do with updating AS to version 2 (2.1.2 is the current stable version I have), and there seems to be an issue with the Gradle (which is part of the whole Android Studio thing that is fuzzy for me). Two; that the file 'ftc.debug.keystore' needs to be in the folder ftcRobotController\libs (which, when I looked, it was already there, so I didn't need to download, phew), And thirdly; that the line of code: compile files('libs/android-support-v4.jar') needs to be deleted. From where, however, they did not say. Here, I just got lucky. Since I know it has something to do with the gradle, I randomly opened the files: build.gradle in the Gradle scrips section of Android Studio and found that line of code. It was located in several places. The build.common.gradle  as well as the build.gradle for each of the module ftcRobotController and Swerve.

Sorry this post got so long, but I wanted to make sure anyone having the same issue, has a little clearer picture of what they might need to do.

rgatkinson, 2016-06-20:
Apologies to all, but we haven't had the opportunity to update the master branch on Swerve Library since mid-March, and, as some have observed, recent Android Studio updates have necessitated that we do so. We hope to get that taken care of by the end of the week; with luck, sooner.

rgatkinson, 2016-06-25:
Fixed in Swerve Library 1.5.3.

closed 2016-06-25 by rgatkinson